fehlfarbe/arduino-motorfocus #2: Buttons not worked for me, here is my solution.

franky384, 2018-12-28:
First of all, thank you for sharing your code,
but the push buttons never worked for me. So after some days of searching and trial and error i finally found a working solution.
Maybe you can have a look at it ?

[image]

[sketch_dec28a.html.pdf](https://github.com/fehlfarbe/arduino-motorfocus/files/2715402/sketch_dec28a.html.pdf)

franky384, 2018-12-28:
I have tried it like in that example, but it hadn't worked for me. I can try it again with my changed code.

fehlfarbe, 2019-02-17:
Should be fixed now. I activated the pullup resistors for button input and checked for BUTTON == LOW. So the wiring in your first post is correct.

closed 2020-12-21 by fehlfarbe