fehlfarbe/arduino-motorfocus #4: Not anymore compatible to Moonlight SW

energiebrand, 2020-12-14:
Hi,
Build the controller using an Arduino NANO clone (FTDI chip), ULN2003 stepper druver and the gear stepper motor. The sketch works out of the box when using the manual control via the buttons.

Just for fun I tried to use the original Moonlight non-ASCOM controller SW but unfortunately it only connects, read the temperature and actual step value and does nothing. The Moonlight SW hangs up (WIN 10 and WIN7). When connecting via PUTTY serial, everything works fine. All listed commands are working.

Then tried the Moonlight ASCOM driver. The driver is recognized by CCDCiel SW for example. Then I open an ALPACA driver and this opened the Moonlight ASCOM driver. But this ASCOM driver does not recognize the Adrduino (sketch).

So I think maybe Moonlight changed something in their SW (Non-ASCOM controller App as well as Ascom driver) so that your sketch is non more supported. Could that be? Did you try to to connect with the latest Moonlight SW?
Best regards and however thanks for the Sketch!

fehlfarbe, 2020-12-14:
Hi,

to be honest I tested the setup only with the INDI moonlite driver. But many people asked in the last weeks if it will run on Windows with ASCOM. So I set up a Win10 VM and tested the "Moonlite Single Focuser" which worked like expected (just with my nano and the motor on a breadboard)

[image]

Then I tried the Moonlite DRO ASCOM driver but it can't even find the COM port:
[image]
Seems like the driver opens all COM ports, sends a command and waits for the right response. Unfortunately the Arduino Nano resets when the COM port is opened and it takes 200ms or so before it's ready to receive commands which takes too long for the Moonlite driver.
You can turn off the auto reset with a 10nF capacitor between RST and GND (I tried a 47nF which is also working). 
Then open the properties windows again and the Arduino will show up. If not, close the windows, press the RESET button on the Nano and try again. It worked on the second try with my setup.
[image]

[image]

Let me know if it's working, so I can update the readme :smiley:

energiebrand, 2020-12-15:
Hi and thanks a lot for fast response!

The solution with capacitor / Reset seems to be the right way but I have only partly success.
Firstly after connecting a 33nF C the Arduino connects with the Dual Focuser Moonlight SW and I was able to control the motor as expected. Very good!

Then I connected via ASCOM using the Moonlight focuser driver and CCDCiel SW. Now it found the Arduino as a Focuser. Very good!

Then I checked with the Single Focuser Moonlight SW on a WIn7 PC but no success...

Then went back to the WIN10 PC and tried again with the Dual Focuser Moonlight SW...no success anymore!!! I tried with several C´s (15nF, 33nF, 22nF, 1mF, new boot of the PC, Reset manually of the Arduino, no chance anymore. The Arduino always connects in the first or second attempt, send some data (Pos and temperature) and then disconnects. Tried it dozend of times.

After that back to the ASCOM driver and always success. I removed the USB, no focuser, reconnceted the USB and success with the ASCOM driver.

So no luck anymore with the non-ASCOM driver. (I cannot use the Single driver because I deleted the SW in a not correct way and it seems that somewhere parts are remaining. This results in that I cannot anymore install the single focus driver...)

To summerize:
- with Moonlight ASCOM driver working on a WIN10 installation using a 15nF, 33nF, 22nF, 1microF
- unfortunately I cannot do more things with CCDCiel because I do not have a camera, but I can see that there is a communication between the Arduino and the PC (RX and TX LED flashing)
- no permanent connection anymore with Dual focuser SW, just a short data exchange
- also it is not possible to open the ASCOm device hub, but maybe I am doing wrong or anything more is needed

So what to do? Since currently I am only using the "Manual button" function I can live with that.
Also later on when I will go to astrophotographie the ASCOM driver works. 
Only limitation for me is the non-Ascom SW.

However the capacitor is working and helps. The other problems seems to be a problem of the used PC and the Arduino clone version.

So no problem with your sketch and I think you should close this issue but add some information. I would recommend to add the C in the description and schematic!

Thanks a lot and if you have any further idea you are welcome!

[image]

[image]
[image]
[image]
[image]

energiebrand, 2020-12-15:
Tried the whole day to get more about the problems I have...

Maybe a 1microF is a better solution. I noticed that the ASCOM driver always recognized the Arduino when the USB cable was connected (...and did not recognize when USB was open), but in many cases there was no communication between the Arduino and the ASCOM driver after the connection was established. So I changed to a 1microF C and RX/TX is flashing constantly. 
So that means even if the ASCOM driver windows shows that is has found a COM port and a focuser is detected there could be that there is no communication.

In addition meanwhile I got the Single Focuser Moonlight SW running again but same problem as I had without the C. It connects now stable, sends some information (Temp and Pos.) and after that it freezes...

fehlfarbe, 2020-12-17:
Sorry my bad, I used a 47**µ**F capacitor. I also found different hints/solutions:
- Update the bootloader: the Arduino clones from China have the old bootloader that takes 1-2s too boot up. I flashed the newer Optiboot with [MiniCore](https://github.com/MCUdude/MiniCore) that boots much faster. At least the setup finds the focuser without the capacitor. But I still can't connect with ASCOM Device Hub.
- Remove the 100nF capacitor on the bottom as described here: https://www.astroscopic.com/blog/disable-arduinos-auto-reset-connection. After that you have manually press the reset button if you want to upload a new firmware
- Does it also freeze if you don't connect the temperature sensor? I just wired the motor but no sensor. Maybe the Arduino sends some bad numbers and the driver freezes?
- Sometimes I have to press the "connect" button twice on the SingleFocuser but after that it is stable
- The DualFocuser looses the connection after some seconds. I assume that happens because it sends an unknown command (starting with a number), doesn't get a proper answer and runs into a timeout:
  ```
  [01:01:21:086] Got new command: GT␍␊
  [01:01:21:108] GT:␍␊
  [01:01:21:114] GT␍␊
  [01:01:21:213] temperature: -127.00␍␊
  [01:01:21:267] Got new command: GP␍␊
  [01:01:21:290] GP:␍␊
  [01:01:21:296] GP␍␊
  [01:01:21:301] current motor position: 0x0820 = 2080␍␊
  [01:01:21:345] Got new command: 2GP␍␊
  [01:01:21:365] 2G:P␍␊
  [01:01:21:372] 2GP␍␊
  ```
  I just updated the code so it will ignore the leading "2" and just parses the command. So motor1 and motor2 show the same values. After that the DualFocsuer doesnt disconnect but freezes :laughing: but SingleFocuser still runs fine.

energiebrand, 2020-12-17:
Thanks  a lot for the deep-dive into the problem!
Yes I have seen in an other discussion that someone has used a much bigger C and thats why I tested also with 22 and 47microF. However a 1microF does also the job and connects stable to the Single Manual Focuser SW.

And yes my NANO clone have the old bootloader.

I do not use the temperature sensor, so the pin is open in all my tries. It shows 0° but still freezes with the Single Manual Focuser SW after a first and short data transfer. It changes the display from "No data" to "0" and the temperature to 0°.
I had only one attempt 3 days ago where it works and it was possible to control the motor 100% and I think this was by random.

Using the Dual Focuser SW: the Arduino connects also, shows the 0 values on both rows but than disconnects after 5 seconds like yours, but without your sketch modification!

Now the bad thing for you... :-(  yesterday I completed my installation and closed the cover. As I said currently I will use the focuser just manually with the buttons in a hand-box I am printing at this moment. 
So with this installation I will not going on to flash a new bootloader. However it would be very interesting to continue and see the results. Unfortunately at the moment I don't have another NANO for more testing.

But it could be that when changing to astrophotography I will use your solution again and need a SW a (ASCOM) controlled focuser. Then I will come back to your proposals.

But regarding the ASCOM driver. Like you my NANO will not really run with it. It connects and most time I see a continously data transfer (RX/TX LED flashing) but no chance to control the motor. Maybe it is the missing astro camera maybe it will not work anyway.

So from my side I will stop here with the current installation but will answer here if I got new results with another NANO I will buy in the future. Thanks a lot for your works! Well I think think discussion will help other people if they are running into similar problems. Attached some pictures of my Newton and the focuser. :-)

[image]

[image]

fehlfarbe, 2020-12-21:
Nice prints! I added your Thingiverse CAD models to the readme :smiley: 

If you want to do further investigations you can enable trace/logging in ASCOM and maybe find an error in the logfile and connect a serial/ttl converter to GPIO 10 and see your Nano's debug output (https://github.com/fehlfarbe/arduino-motorfocus#other-ascom-errors)

I will close the issue for now but feel free to reopen :)

closed 2020-12-21 by fehlfarbe