fehlfarbe/arduino-motorfocus #11: Several updates to enable use on Sparkfun Pro Micro

jvanderk, 2021-12-07:
Trying to use the code on a Sparkfun Pro Micro clone, I ran into several issues:

1. serialEvent was not called automatically after loop().
2. SoftwareSerial had a compilation issue related to the assembly in there.
3. An issue with temperature string formatting.

Made changes to code and added some documentation and a choice to platformio.ini

I'm going to try to submit the changes using a for/pull but I've never done that so bear with me....

jvanderk, 2021-12-07:
I think I created a pull request but not sure how to link that to this issue.

fehlfarbe, 2021-12-07:
Thanks! I will take a look at it tomorrow :)

fehlfarbe, 2021-12-08:
I think I can't push my changes on your pull request but I created a new branch `promicro` with your changes and added some new environments for different configs to platformio.ini. There is a patched version of SoftwareSerial on Github, so I'm using this repo for SoftwareSerial so other users don't have to modify the lib.

Can you take a look at it?

jvanderk, 2021-12-08:
Cool!

Tested 

```
# Sparkfun ProMicro
[env:promicro16]
[env:promicro16_debug]
```

environments and both seem to work (manual buttons and ekos controlled).

Patched version of SoftwareSerial is OK.

Still in a maze on the pull request workflow :--)

closed 2023-06-08 by jvanderk