fehlfarbe/arduino-motorfocus #13: I'm stuck at Arduino flash

zeltragh, 2022-02-01:
Hello,

I try to flash my arduino nano with the arduino-motofocus project but error comes.

[image]

i don't know i did bad... 

Can you help me ?

zeltragh, 2022-02-01:
[image]

fehlfarbe, 2022-02-01:
Hi!

Open a terminal in VSCode and type

`pio platform install atmelavr`

This should install the needed platform. If this doesn't work, please add a screenshot or log output here :)

zeltragh, 2022-02-06:
Thank you for help.
Finally it was pio platform issue. After reinstall of this , problem solved.

closed 2022-02-06 by zeltragh