fehlfarbe/arduino-motorfocus #16: Focuser is initializing in moving motor state on connection

cinosh07, 2022-06-22:
I'm using 28BYJ-48 stepper motor with ULN2003A breakout board and when I connect to the focuser, It always start in moving state to true. I have to go in the device hub, make a small move, then the moving state change to stop

Here is my log
[ASCOM.MoonLite.txt](https://github.com/fehlfarbe/arduino-motorfocus/files/8959026/ASCOM.MoonLite.txt)

cybye, 2022-10-03:
#15 <- please check if this helps

cinosh07, 2022-10-03:
> #15 <- please check if this helps

Thanks for your reply, That's the first thing I've tried. At first I was thinking that it was doing that on only 1 of my 2 focuser. But finally it does it on both. On with 5v motor and the other one with the A4388 driver and 12 volt. Both does the same thing. But I didn't change the bootloader. That the original with capacitor.