fehlfarbe/arduino-motorfocus #17: wrong wiring for A4988 ?

Holgervisitinggithub, 2022-08-13:
Disclaimer: I am no expert at all, but at least it worked for me.

Picture of the wiring from A4988 to the arduino seems to be wrong.
The "DIR" pin from A4988 needs to be connected with D3 instead of D5
The "Step" pin needs to be connected with D5 instead of D6

according to the code
#define PIN_DRIVER_DIR 3
#define PIN_DRIVER_STEP 5

Thank you for the great work!
[image]
!

Ardufan, 2022-08-18:
I noticed as well that the wiring and code are inconsistent. 

And I would recommend to take the chance to rename this piece of code:

#define PIN_DRIVER_ENABLE 4

The pin n. 4 (arduino side) is connected to the SLEEP and not the ENABLE pin on the driver side. It would be advisable for code clarity to rename this #define as :

#define PIN_DRIVER_SLEEP 4

and change accordingly this line in setup(): 

stepper.setEnablePin(PIN_DRIVER_ENABLE);

to 

stepper.setEnablePin(PIN_DRIVER_SLEEP);

fehlfarbe, 2022-08-22:
Hey thanks! I updated the picture and changed the name to `PIN_DRIVER_SLEEP` :+1: (You may have to clear your cache to see the updated image)

closed 2022-08-22 by fehlfarbe

Ardufan, 2022-08-22:
thank you for sharing your nice code! 

PS: I added to my own copy a TM1637 6 digit display, just to show the current position also when operations are manual.
It only requires 2 arduino DIO pins and the library TM1637TinyDisplay6

commit 3436720b (2022-08-22): Fixes #17