fehlfarbe/arduino-motorfocus #19: Ideas for working around "avrdude: butterfly_recv(): programmer is not responding" ?

frankbraker, 2023-01-23:
When I try to upload the hex file (regardless of which target I select), I get something along the lines of:

Found programmer: Id = "F"; type = @
    Software Version = \xdc.u; Hardware Version = \x93.\xf0
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1

It always fails.

I've confirmed my hardware is based on ATMEL M328P.  It's a nano.

Toboi, 2023-02-01:
I am not sure if this is the same problem I experienced, the output is different.
But there seems to be a version of the Arduino nano with a new bootloader which requires a different board variable (See [platformio docu](https://docs.platformio.org/en/stable/boards/atmelavr/nanoatmega328new.html))

In the platformio.ini I had to change
```
# Arduino Nano
[env:nanoatmega328]
board = nanoatmega328
```

to
```
# Arduino Nano
[env:nanoatmega328]
board = nanoatmega328new
```

fehlfarbe, 2023-02-01:
Good idea @Toboi !

The nano has different bootloader versions that aren't compatible so maybe replacing the board will fix it.

If this doesn't work you could try to upload the compiled hex file manually via terminal: Activate verbose output in the Arduino IDE and upload a sketch. Look for the `avrdude` command that is used to upload the hex file in the output window of the IDE and run the same command with the replaced path for the hex file (should be at `arduino-motorfocus/.pio/build/nanoatmega328/firmware.hex`)

frankbraker, 2023-02-04:
> I am not sure if this is the same problem I experienced, the output is different. But there seems to be a version of the Arduino nano with a new bootloader which requires a different board variable [...]

[image]

This worked for me!!  Thanks!!

closed 2023-02-04 by frankbraker