fehlfarbe/arduino-motorfocus #20: Do the buttons work for you?

frankbraker, 2023-02-06:
I was certain there was something wrong with my hardware - after trying several times with one I had wirewrapped, and finally using a breadboard and it still seemed like it was dead... and then when the house was as quiet as it could get in the middle of the night, I finally noticed a very very quiet bumping sound coming out of the stepper motor when I pressed the buttons.

I tried changing the # of steps in BTP_STEP from 32 to 4096 and that helped - but the whole thing is still way too slow, I can't see the knob moving at all.

Am I still doing something wrong?  Do the buttons work for you?

I would file this under "tips and tricks for troubleshooting" - and just getting it to make a sound or see it moving seems like a good first step.

fehlfarbe, 2023-02-06:
Do you use the 28BYJ-48 stepper motor with ULN2003A breakout board or a different driver like the A4988?

The ULN2003A breakout board works for me out of the box with a passable speed:

https://user-images.githubusercontent.com/3355380/217092530-a81abef7-8d06-46c4-a3c4-9a8a5a7ba983.mp4

frankbraker, 2023-02-07:
I'm using the 4988 (shown at the bottom of your breadboard).  Because I'm using this with a Moonlite focuser, I think the stepper motor on the focuser is geared way down to go pretty slow even when it's full stepping.  I imagine I'll want it to go as fast as it possibly can when using the buttons, and only when using the ASCOM driver for fine control it might be more appropriate to go slower then (albeit, probably not necessary to go slower).

fehlfarbe, 2023-02-07:
Just to be sure: I tried with the A4988 driver and a random stepper motor. The steps/speed is quite fast/ok.

https://user-images.githubusercontent.com/3355380/217333119-50e2b766-fd25-47c9-8771-9daf9c98cf68.mp4

So I think it's the reduction gearbox. Maybe it helps when you increase BTP_STEP and speedMult (that limits the maximum speed for movements)
https://github.com/fehlfarbe/arduino-motorfocus/blob/4e0ffb6a2911d06b1b8edb75302927f77b042a93/src/arduino-motorfocus.cpp#L45
and increase the acceleration rate:
https://github.com/fehlfarbe/arduino-motorfocus/blob/4e0ffb6a2911d06b1b8edb75302927f77b042a93/src/arduino-motorfocus.cpp#L80

frankbraker, 2023-03-13:
This issue can be closed as far as I'm concerned.

Thanks so much for all the help!  Sorry I didn't get back sooner, but I've been away.

I was finally able to get mine working properly.  The labels on the A4988 confused me, where 1A, 1B, 2B, 2A should be paired as 1A with 2A and 1B with 2B.  But that's not all - because I'm using a Moonlite stepper motor it's a unipolar, which means don't connect the 12V line (only connect the 1+, 1-, 2+ and 2- lines).  See page 8 of http://www.indilib.org/media/kunena/attachments/1/HighResSteppermotor107.pdf and OP of this thread https://www.indilib.org/forum/general/1260-controlling-moonlite-focuser-from-raspberry-pi.html for details.

After working through these I finally have things moving, and I set the current limiting Vref to 0.24v to facilitate the 0.15A the motor can handle, using S1=0.2 (source http://hiletgo.com/ProductDetail/1952643.html ).  R10=20,000, so Vref(target current) = 0.15A * (8* 0.2) = 0.24V  -- see https://www.instructables.com/Drive-a-Stepper-Motor-with-an-Arduino-and-a-A4988-/ and notice that any "0.8" magic number is really 8*S1.  Hopefully this helps someone.

This still didn't work well with whole steps (missed a lot of steps even with BTN_STEP=32 - but works okay with 1/8 stepping or even 1/4 stepping.

closed 2023-03-13 by frankbraker