fehlfarbe/arduino-motorfocus #21: Nema17 with A4988

HansBeg, 2023-02-19:
Hallo and many thanks to share this work with us. After 35 years without electronic bricolage you invited me to follow you. Thanks! I have Win10, the Nano and the 28BYJ-48 stepper motor with ULN2003A breakout board, with the temperature sensor wired. With the buttons it works out of the box. The „nanoatmega328new“-issue I found out by myself by hazard.
But, I have two issues/ problems.

At first „ASCOM cant find the focuser“:
After using an 0,1µF capacitor I got the connection! I tried also the way with „MiniCore Bootloader“. I used the master „Nano as ISP“. Than I choose ATmega328 offered by MiniCore (in Sketchbook), see Pic. But „Burn Bootloader“ doesn’t happen. When I select the Arduino Nano board I burned the bootloader. And now I can  reach the setup, and then? How can I operate with the Focuser (hardware)? Moonlite Single Focus find the focuser, set the temperature every 0.5sec, but no movement is possible. I found an other Application: AP Mini 19.12-1. This works fine with temperature – but as non ASCOM device, as a simple MoonLite. When I choose „ascom“ the motor never stops. I can live with AP Mini, this is ok for me.

At second: I need the bigger motor, an Nema 17 with the A4988 driver. I made the wiring on the breadboard, using MB102. I checked out before using, that the 12V goes to the + and – rails. The 5V I got via USB. Than I wired, switched „on“, the yellow light blinked  few times and the voltage converter for 5V said good bye. The motor doesn’t moved - I know the motor need too much current for the MB102. I checked the data sheet which Az-Delivery offers. I found: 

„Connect RST pin to the SLEEP pin to keep the driver enabled (Black wire).
REMEMBER to put a large 100μF decoupling electrolytic capacitor across motor
power supply pins, as close to the board as possible, like on connection diagram“

But in the wiring diagram of the project I didn’t find such need. And what is with the connection between RST and Sleep? Did you tested the combination A4988 with a Nema motor? Is the wiring in the project right?  (I read the closed request #17)

regards Hans

[image]
[image]
[image]

fehlfarbe, 2023-02-22:
Hi Hans,

I will try to take a deeper look into your problems this week.

Just for now: 

I tested a Nema motor with A4988 driver some weeks ago -> https://github.com/fehlfarbe/arduino-motorfocus/issues/20#issuecomment-1421269295
I connected the 12V of my bench power supply directly to the A4988 driver. Did you connect the 12V to MB102 and the 5V output of the MB102 to the driver? I think the motor draws too much current for the small 12V -> 5V linear converters of the MB102 so just connect the 12V directly to the driver VMOT / GND as suggested here **and check that you don't mix it up with the 5V / VDD pins**! You can also add the 100µF to the motor power supply lines. I didn't need them for my testes but if the motor has some load they might be required.
[image]
https://www.pololu.com/product/1182

HansBeg, 2023-02-22:
Hi, 
yes, the burned MB102 was my fault. Learn by errors.
I loaded the actually code - without success. Then I added in line40 also the "new" and the Nano flashed (without an capacitor btw GND and RST. The Nema 17 (with Elko) turns with the buttons and with AP Mini after the second try.  Uff! 
The motor works loud and hard. I will use microsteps. On the A4988 are the MS1, 2 and 3. In the data sheet I see this.

[image]

Does it means High: 5V and Low: GND? Or should I set D9 and D10 and connect?

kind regards to Dresden, my favorite town since years.
Hans

fehlfarbe, 2023-02-26:
Hi,

so it's working now? Does the motor rotate or does it just "vibrate"? You can change the maximum current with the small potentiometer on the driver. Maybe it doesn't get enough current?!

Indeed, LOW means connected to GND and HIGH means connected to 5V. The pins have internal pulldown resistors so if they are unconnected they are pulled down to GND.

And don't forget to connect RST to 5V and SLEEP  to D4. The Nano sets the sleep pin after some seconds to disable the driver and save energy and disables sleep when a move command or button press was detected. If you don't want/need this behavior you can connect RST to SLEEP and D4 stays disconnected.

D9 and D10 are only pins for debug reasons. You can connect a USB-TTL device to get the debug messages printed by "debugSerial.print()" functions. Just leave them unconnected.

I'm checking the Arduino firmware with some programs on Windows and found some little bugs that I'm fixing at the moment. Where did you download the "AP Focus MoonLite" program?

Best regards from Dresden,
Marcus

HansBeg, 2023-02-27:
Thank's for the answer. The motor turns, but not smooth. Since a few steps it is a pause. But, when I say 10 turns (in AP Focus) I get 10 turns. In the web is this behavior described.
The microsteps are now ok. 
The download page I hadn't noticed. But I found it. Here you are: https://miltonhill.us/software/index.html 
[image]
Meanwhile I tried out the Moonlite ascom (Build: 1.3.1.0) with little success. I can read the temperature. After reset of the nano the motor fliged (zappelt) but  I can't do anything to turn some steps.
many regards from Reutlingen

HansBeg, 2023-02-27:
P.S.: sometimes (after 1min or so) happens one given funktion, e.g. "HOME"
The whole time TX and RX are blinking. Temperature is always up to date (change after 1 or 2 secs when neccessary)
Hans

fehlfarbe, 2023-02-27:
Thanks for the feedback. I updated the code and tested with APT (Astro Photography Tool) and the AP Mini tool and they worked for me. You dont' have to modify the config again. I added the `nanoatmega328new_A4988driver` config that you can change on the bottom of VSCode:

[image]

HansBeg, 2023-03-01:
Hallo and many thanks for your labor. I updated my software. I works. I realized, that only full steps and 1/4 steps are possible. I don't know why. But that's good and enough for me. Now I have to fill this in a appropriate box, respectively on a conductor plate. 
I wish you the very best for the future! 
Bye

closed 2023-03-01 by HansBeg

fehlfarbe, 2023-03-03:
Nice! Feel free to open a new issue if there any problems :smile: