fehlfarbe/arduino-motorfocus #23: Upon connecting, the red light blinks 100011000 at about 0.5Hz until I disconnect

frankbraker, 2023-08-20:
[image]

When I connect from N.I.N.A, Windows claims it connected, but the red LED on Arduino blinks 1 on, off for about a second, then 2 rapid ons, and off for about another second.

[ASCOM.MoonLite.2015.119360.txt](https://github.com/fehlfarbe/arduino-motorfocus/files/12387055/ASCOM.MoonLite.2015.119360.txt)

fehlfarbe, 2023-08-28:
Hey, sorry for my late response. The builtin LED turns on when the motor is moving but it seems that the target position and the current position is the same in your screenshot even though "is moving" is set to true :thinking: 

The connection seems to be ok because NINA sends commands and gets responses but you can't move the motor with the buttons?

frankbraker, 2023-08-28:
Correct, both the buttons are unresponsive, and there is nothing that can be done via the driver in NINA.

fehlfarbe, 2023-08-29:
Might be a bug for the GI ("is motor running?") command. Can you try this branch: https://github.com/fehlfarbe/arduino-motorfocus/tree/fixrunning ?
I tested it with NINA in my Windows VM and it seems to work.

Also the LED should only turn on when the motor is moving.

frankbraker, 2023-08-30:
Well there are 2 lights: the amber one lights up (but fails to move) when I press a button.  It's the red light that is doing the blinky pattern.

I'll try the branch and let you know if it behaves differently.

frankbraker, 2023-09-17:
Now even after I remove the 10uF cap from reset it doesn't program:
[image]

Now if I revert to the old stuff that worked before for me, and I select the atmega328new_A988driver - it builds and does upload successfully - so there's something that has broken in the latest branch.  When I try to build that it gives an error "no id" or something like that.

[image]

AND THEN, for no explainable reason, trying it again on your latest commit on your branch and it suddenly does upload!  No idea why, when it failed at least 4 times before.

[image]

I'll try this now to see how much better it works (if at all).

frankbraker, 2023-09-17:
Yes it connects and sort of works.  It is way too slow, and I imagine it might be best to at least half, if not quarter step to insure steps are never ever missed, but the problem that opened this bug seems to be addressed now and I'm able to send commands over NINA.  Thanks for the help!

frankbraker, 2023-09-17:
Unless anybody wants to reopen this I'm marking it closed.

closed 2023-09-17 by frankbraker