fehlfarbe/arduino-motorfocus #24: 28BYJ-48 stepper motor idle issue

asahiGR, 2023-08-30:
Hello,

when there is no action and the motor is idle, it continuously clicks and get fairly warm after a while. The led on the driver pcb rotate showing it's active but not moving. Any way to stop it if there's no input? Some addition to the code?

fehlfarbe, 2023-08-31:
Hi!

As you mentioned before, the motor driver is turned off if there was no movement for 15s (millisDisableDelay) and there shouldn't be a difference if the focuser is connected to a PC, a power bank or 5V power supply.

If the driver LEDs are blinking the controller sends movement signal to the driver or a loose connection or some similar hardware problem creates signals? Did you connect the GND from the driver board to the GND of the nano? I just tried with my 28BYJ-48 setup and the LEDs turn off after the 15s delay - even when working with a power bank.

asahiGR, 2023-09-04:
Hi, thanks for you feedback. The motor works good, i've installed buttons and work it in two speeds. Everything runs smooth. Only problem, it does not turn off. All wiring is correct, gnd and power supply is common with arduino power supply. I flashed it again with nanoatmega328  config, same issue. I am using arduino nano clone CH340 in case it matters.

nettylink, 2024-01-22:
Hi,
I've seen the same behavoir, when using the buttons and the potentiometer is completly open / at "full speed".
After decrasing the speed with the potentiometer a tiny bit I've not seen this issue anymore.