fehlfarbe/arduino-motorfocus #25: Changes to Readme to explain where to change time/step, step resolution, etc.

frankbraker, 2023-09-17:
So things look really great right now - in that they seem like they are within reach.

I am sort of confident that I could try to manually change the time/step and the step resolution and get my hardware to work with the moonlite focuser I have.

However, I'd really like to follow the directions on the tin, and right now there isn't a lot of detail in this domain.

When I just build what is at the top of the latest branch, and run it, I believe I had hard-wired my driver to use 1/2 steps, or maybe 1/4 steps.  So that might be a problem for starters.

If I change nothing, I find that the Moonlite ASCOM driver limits how many steps can be traversed by 10,000.  But if I try to move 10,000 steps, it doesn't move very far - less than a millimeter.  This means I probably got a wire lose to the coils, or backwards.  But it's strange because I can use the buttons and they seem to work - or at least they did work fine at one point.  Regardless of all this, it would be really nice to have it pointed out in the README file where to look in the code for fine tuning the firmware to match the configuration of hardware in use.

fehlfarbe, 2023-09-19:
Hi!
Sorry, I had too much work to do the last days. Maybe I will find some time this week to add some more tips for beginners to the readme and move some default values to macros. If you leave the MSx pins unconnected on the driver it should already work in full step configuration:

[image]
(I will add a link to the [Polulu doc](https://www.pololu.com/product/1182))

At least I finished my focuser dev board so I don't have to fiddle around with loose breadboard connections anymore :smiley: 
[image]

I'm also planning to implement the arduino web uploader ( https://github.com/dbuezas/arduino-web-uploader ) for this project so beginners don't have to install platformio and can flash their Arduino via browser.

frankbraker, 2023-09-24:
Those are okay, and thanks for trying to answer my question.

What I was really asking for are the variables in the code that could be better documented.  Right now I'm just guessing at what is okay to change and what I would be better off leaving alone.