fehlfarbe/arduino-motorfocus #27: Cnc shield arduino nano

hpacone, 2024-01-19:
Hallo, erstmal erstmal ein großes Lob an dich und dein Projekt!!
Dann eine riesen bitte! Glaubst du wäre es dir möglich das ganze auf einen Arduino Uno umzusetzen?? Best greetings from Austria ✌️

fehlfarbe, 2024-01-20:
Hi,

you can also write in German if that's easier for you :)

I added a version for Arduino Uno and CNC shield in the [arduino_uno_cnc_shield](https://github.com/fehlfarbe/arduino-motorfocus/tree/arduino_uno_cnc_shield) branch. Just checkout the new branch and try to flash your Uno with this version. I'm controlling the X axis motor. Unfortunately I don't have a CNC shield at home so I cannot test the changes. If it doesn't work, I will order a shield from eBay to test the code.

hpacone, 2024-01-21:
Hallo,
das ist wirklich sehr nett von dir ich danke dir vielmals!!! Ich werde das die Tage testen und dir ein Feedback geben! Dankeschön
Glg

fehlfarbe, 2024-01-23:
Hey, ich habe gerade noch mal ein paar Anpassungen vorgenommen. Leider wurden meine Pinänderungen im Code wieder überschrieben. Da mein CNC-Shield vorhin ankam, konnte ich es auch gleich testen. Bei meinem Motorkabel sind vermutlich die Pins vertauscht, daher springt er gerade nur hin und her. Ich hoffe, bei dir klappt das.

Die Buttons zum rein/raus zoomen sind auf die Z/Y Endstops gelegt. Damit kannst du es auch probieren.

hpacone, 2024-01-23:
Danke für deine Bemühungen! Ich werde das morgen gleich probieren. ✌️

hpacone, 2024-02-02:
HI, leider finde ich keine Eintrag zu flashen eines Unos in der Auswahlliste!? 
Ich finde auch bei den env keinen Eintrag zum uno cnc shield.
Danke Sg

fehlfarbe, 2024-02-02:
Hast du den richtigen git branch ausgecheckt?

```
git pull
git checkout arduino_uno_cnc_shield
```