fehlfarbe/arduino-motorfocus #28: DS18B20 temperature sensor no "negative values"

nettylink, 2024-01-22:
Hi,

I'm using the arduino-motorfocus with N.I.N.A. and have noticed, that negative temperatures are not shown correctly.
With negative temperatures always 0.00° are shown.

The current code always returns "0#" on negative temperature readings of the DS18B20.

I've change the following code from:
```
    byte t_int = (byte)temperature << 1;
    t_int += round(temperature - (byte)temperature);
    Serial.print(t_int, HEX);
    Serial.print('#');
```
to:
```
    int t_int = (int) temperature;
    int tmp = (t_int * 2);
    char tempString[8];
    sprintf(tempString, "%04X", (int) tmp);
    Serial.print(tempString);
    Serial.print('#');
```
I guess there are better solutions, but with this code negative temperatures are shown correctly.

fehlfarbe, 2024-01-24:
Thanks! I tested several values and simplified the code a bit :)

closed 2024-01-24 by fehlfarbe

nettylink, 2024-01-30:
Hi,

thank you very much for the new code.
I've also tested a bit more.

When duplicating the temperature as integer, we get a 1 degree resulution.
```
int16_t t_int = (int16_t)(temperature*2);
```

When duplicating the temperature as a float, we get a 0.5 degree temp resulution.
```
float d_temperature = (float)(temperature*2);
int16_t t_int = (int16_t)(d_temperature);
```
Here are few test values.
| temperature (from DS18B20) | tempString (old) | tempString (old as int) | in N.I.N.A. (old) | tempString (new) | tempString (new as int) | in N.I.N.A. (new) |
| --- | --- | --- | --- | --- | --- |  --- | 
| 4.49   | 0008     | 8   | 4.00   | 0008     | 8   | 4.00 | 
| 4.51   | 0008     | 8   | 4.00   | 0009     | 9   | 4.50 |
| \-4.49 | FFFFFFF8 | \-8 | \-4.00 | FFFFFFF8 | \-8 | \-4.00 |
| \-4.51 | FFFFFFF8 | \-8 | \-4.00 | FFFFFFF7 | \-9 | \-4.50 |

fehlfarbe, 2024-02-01:
Ok that's weird because the temperature is duplicated as float in the parenthesis and afterwards casted to integer so it should be the same :thinking: 

```
(int16_t)(temperature*2) == (int16_t)((float)(temperature*2))
```