fehlfarbe/arduino-motorfocus #30: How do I Install Minicore Bootloader?

NovaAstrophotos, 2024-05-09:
I don't have any knowledge about programming or Arduino, but I liked your project and decided to make a auto focuser for my telescope, I managed to insert your code through Arduino, but the Moonlite didn't recognize it, so I would like to know how to insert the minicore bootloader into your code.

I'm using the cheap Arduino Nano clone called:

Nano 3.0 with the bootloader for arduino CH340 USB driver 16Mhz ATMEGA328P and a ULN2003 board with 28BYJ-48 5V 4-Phase Stepper from aliexpress.

I would like to know if you could make a mini tutorial teaching how to inject the Mini Core bootloader into your project, or recommend a tutorial so I can learn more.

fehlfarbe, 2024-05-09:
Hi,
I assume you are using Windows? Sometimes flashing the Mini Core bootloader isn't enough. I would suggest just to install a 10µF or 33µF capacitor between RST and GND. This prevents the Nano from reboot when the COM port is opened.

NovaAstrophotos, 2024-05-09:
Hi, I managed to install the minicore bootloader, but even so Moonlite I couldn't find it, I'm going to buy some capacitors, which is the correct side that I have to install the capacitor on?, my Arduino has RST and GND on both sides. thanks in advance

fehlfarbe, 2024-05-14:
The side shouldn't matter. Afaik both pins are connected to the same pin on the microcontroller.