fehlfarbe/arduino-motorfocus #32: The capacitor trick has an issue

pulsar123, 2025-03-29:
First of all, thanks for this project! 

I have an issue with the capacitor trick - my ASCOM driver couldn't connect to the focuser until I soldered 10 uF capacitor between RST and GND of Arduino Nano. Now it connects all right, but I have another issue - I can no longer reprogram Arduino. (Arduino IDE fails to connect.) Is it a known issue? May be changing the value of the add-on capacitor can help? Or the only solution is to solder/desolder the capacitor every time I need to reprogram the board?

I also have a question about the meaning of the speed parameters: speedFactor and speedFactorRaw. In your code the values for these two are 16 and 4. Do I need to keep these values, and only change SPEED_MULTIPLICATOR  to achieve the speed I need? In my case, I am using 1/32 microsteps, and my target speed and acceleration are 3765 ustep/s and 941 ustep/s^2, respectively. 

Finally, I do not use IN/OUT buttons or a pot. I just set

int btn_in = HIGH;
int btn_out = HIGH;

throughout the code - this should do the trick, correct?

MonsieurFantome, 2025-04-14:
You can program by hitting the reset button at the right time but it's tricky.

fehlfarbe, 2025-04-15:
Hey @pulsar123, sorry for my late replay!

As @MonsieurFantome suggested, you can try to press the button at the right time or maybe solder a switch/button in between like `RST -> 10µF cap -> switch -> GND` so you can use the switch to select "programming mode" or "ASCOM mode".

I think changing the `SPEED_MULTIPLICATOR` to a higher value should  do the job because the `speedFactor`, which is set via ASCOM, is multiplied with this value.

You can set the buttons manually to `HIGH` but this shouldn't be necessary because Arduino's internal pullups are already set to `HIGH`:
```
// setup buttons
pinMode(BTN_IN, INPUT_PULLUP);
pinMode(BTN_OUT, INPUT_PULLUP);
pinMode(BTN_POTI_SPEED, INPUT_PULLUP);
```

pulsar123, 2025-04-15:
Thanks for all replies, I ended up soldering a tiny switch.